mofosyne/arduino-gameboy-printer-emulator #10: _FDEV_SETUP_WRITE is not declared in scope on teensy 4.0

the100guy, 2019-11-24:
Hello, I'm experiencing a bit of an issue here...

Trying to upload the code to a Teensy 4.0 results in the following error:
`gbp_emulator:676: error: '_FDEV_SETUP_WRITE' was not declared in this scope
   fdev_setup_stream (&serialout, serial_putchar, NULL, _FDEV_SETUP_WRITE);
                                                        ^
gbp_emulator:676: error: 'fdev_setup_stream' was not declared in this scope
   fdev_setup_stream (&serialout, serial_putchar, NULL, _FDEV_SETUP_WRITE);
                                                                         ^
'_FDEV_SETUP_WRITE' was not declared in this scope`

Keep in mind, I'm quite new to this Arduino stuff and would appreciate any help I can get. I have looked elsewhere for solutions (such as including libraries) but they aren't really working out, unfortunately.

mofosyne, 2019-11-26:
Hmmmm... I've never tried Teensy before. This was tested on a arduino nano. Don't have a teensy on hand unfortunately. Happy to take pull requests as long as it doesn't break support for the cheaper arduino nano.

mofosyne, 2019-11-26:
This has some information about the problem and maybe a solution for teensy.

https://forum.pjrc.com/threads/24730-fprintf-with-Teensy3-1-hangs

The current code is failing in https://github.com/mofosyne/arduino-gameboy-printer-emulator/blob/master/gbp_emulator/gbp_emulator.ino#L676

The reason I had that line is to support printf in arduino.

the100guy, 2019-11-30:
I read around a bit and it seems that the teensy has weird printf support. I ended up just purchasing a Nano later on which I'l be using for pulling the pictures off of my GB Camera. If I happen to get it working on the teensy, though, I'll let you know. Thanks anyway!

closed 2019-11-30 by the100guy

mofosyne, 2020-04-30:
Did it work for you?