mofosyne/arduino-gameboy-printer-emulator #11: Beetle USB support

eried, 2020-03-14:
Hi! This is awesome, I got it working on a micro without issues, however I was trying to make it work on a Beetle (like a Digispark but with 32u4) but the Gameboy does not sees the printer :/ , which pins would be your recommended replacements for the original ones?: 

[image]

Is the interrupt pin the only requeriment?

mofosyne, 2020-04-29:
Yes the interrupt pin is the only requirement, since this is manually checking the edge of each clock for each bit.

eried, 2020-04-29:
:) it worked fine, I ended trying different pins until it worked

twitter.com/eried/status/1238774188623699969

closed 2020-04-29 by eried

mofosyne, 2020-04-30:
Awesome, maybe tell me what pins worked and I'll add it to the instructions.

eried, 2020-04-30:
Oh, actually I checked the code and I used a small leonardo board. Not the beetle at the end