mofosyne/arduino-gameboy-printer-emulator #13: Can't save canvas to image / split images in batch print

BjornB2, 2020-04-28:
Hi! 

First of all: thanks. Thanks for your great work. See my insta @Gameboycameramaniac for what I've done with your help.

I have 2 requests though:

1. The image renders as a canvas, but I see no simple solution to save the canvas as an image. I made a very basic and ugly work around for it by rendering the canvas to an image, but maybe you can do this in a more professional way 😊 See [Archief.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/4547695/Archief.zip)

2. Would it be possible to batch print on the gameboy camera, and then split into separate files in the decoder? This would reduce the amount of time to dump 30 photo's very much.

Thanks!
Björn

mofosyne, 2020-04-29:
On point 1, have you tried right clicking and saving the canvas in firefox/chrome? On point 2, it should be possible, if you look at the output, I print out the status of each control message and you can find deliminators for new images via it.

Don't have much time these days since I am working full time on this new role as an embedded engineer but I'll be happy to guide you though and accept pull request.

Just email me or give me your telegram ID or something, you can then ping me to do code pulls or code suggestions.

BjornB2, 2020-04-29:
Hi Brian,
Thanks for your answer. On point 1: Indeed, in chrome I can save by rightclicking. In safari this option isn't available. That's why I made the crappy 'render as jpg' option, but it's not really something I would publish since it's a mess.

If you don't have time I will have a look at it, but I am not really experienced with javascript, so it might take a while.

Do you know what happened to GBCam Studio? Also looks like a real nice option. The download link is listed as "coming soon" though.

https://www.maxpiantoni.com/projects/gbcamstudio

For telegram: https://t.me/bjornheirman

Regards
Björn

BjornB2, 2020-04-29:
Hi Brian,

For point 1: I just made a nicer version with a download button and created a pull request for it.
It's the very first time I've ever created a pull request so forgive me if I made a mess of it.

Regards,
Björn

BjornB2, 2020-04-29:
Forget the first pull request. Should have taken the time to figure out how Github works before.
Made a proper pull request now 😉

As for the second part (split into separate images), that would probably mean creating multiple canvasses or something like that? Splitting the input from the textarea is indeed easy, but I'm new to this whole canvas stuff.
On 29 Apr 2020, 13:02 +0200, Björn Heirman <bjorn@heirman.nl>, wrote:
Hi Brian,

For point 1: I just made a nicer version with a download button and created a pull request for it.
It's the very first time I've ever created a pull request so forgive me if I made a mess of it.

Regards,
Björn

BjornB2, 2020-05-02:
Download button added. Batch print maybe later :)

closed 2020-05-02 by BjornB2

mofosyne, 2020-05-03:
Very much appreciated thanks. Credited.

----------

For anyone else who stumbles across this. BjornB2 also added`colors = new Array("#ffffff", "#aaaaaa", "#555555", "#000000");` in this link below that allow for opportunity to change palettes colour as well. One suggestion I made is to allow this to be accessible via dropdown.

https://github.com/mofosyne/arduino-gameboy-printer-emulator/blob/master/jsdecoder/gbp_gameboyprinter2bpp.js#L23