mofosyne/arduino-gameboy-printer-emulator #16: Feature Request: Create a library from this?

HerrZatacke, 2020-05-02:
Hello,
first I have to say thank you for your great work providing this.

I'm currently implementing this on a esp8266 module.
My goal is to provide the "printed" images voer an integrated webserver via WiFi.

As my dev-backgroudn is mainly Web-Frontend (js/css/react...etc) I'm struggling with the Arduino/C in general.

As the esp8266 has internal WiFi (and even sort of an internal filesystem), I guess It should be capable of retrieving/storing the images and provide them via a simple webservice. (or even upload them to some image hosting sites....)

What I'd imagine is being able to do something like this:
``` c
#include "awesomeLibrary"

GBPrinter printer(GBP_SO_PIN, GBP_SI_PIN, GBP_SC_PIN);

void setup() {
  printer.init();
}

void loop() {
  while (printer.available()) {
    byte data = printer.read();
    // do something with the data...
  }
}
```
Your thoughts on this are much appreciated.

mofosyne, 2020-05-03:
Don't have much time, but found someone who managed to treat my code as a lib, so if you can check how he did it in https://github.com/xx0x/gbpxl . Maybe you can add it to this repo and I'll pull it in so others can have an easier time doing so in the future :)

HerrZatacke, 2020-05-03:
Thank's for pointing me in the right direction! This is closer to what I'm looking for.
I'll see if I can work with this and if so will share my insights.

mofosyne, 2020-05-03:
Not a problem. Btw got telegram? I found a few people recently interested in this project this month, I could get you all to talk to eachother and see what you all learn about it. E.g. Working out how to get pokemon to print properly.

HerrZatacke, 2020-05-03:
Yep. I actually do have Telegram. If you share a link to a group I'll join. (guess that's the most practical way)

mofosyne, 2020-05-04:
https://t.me/gameboycamera just created it. Hopefully it works :)

mofosyne, 2020-08-16:
While not exactly a lib, V2 is written in such a way to be easier to port over to other project now.

closed 2020-08-16 by mofosyne