mofosyne/arduino-gameboy-printer-emulator #17: Problem for printing with games other than GB Camera : probable hardware issue

Raphael-Boichot, 2020-05-04:
Hi Brian.

Just to refresh your mind, this is a attempt to solve the issue with your code and games other than GB Camera.

I've tried to write the most minimal Arduino code (so I presume the fastest) to catch the bit flux from Game Boy with Arduino Uno when printing. Basically I just say the magic word 10000001 at the ninth byte of first command, not very impressive, but sufficient to prime a communication stream with game boy.

GB Camera works fine, I got the bytes with the expected format and some data packs (your code is of course the reference for this, no need to go further on my side).

SMB Deluxe and Zelda rapidly present missing bits within the communication frame (as soon as the second byte sometimes). I tried GB Color, Game Boy Advance, GB Boy colour (chinese hack), Zelda regular and chinese hack. The same for any combination : GB Camera prints from all devices, but nothing else. Surprisingly, SMB and Zelda do not mind receiving the magic byte, they send data whatever the message.

I measured the mean time to send a byte (average using the Arduino console timetracking): 1.3 ms for GB Camera, 2 ms for SMBD (!), 0.92 ms for Zelda. This latter is below the time given for DMG Game Boy to send a byte (0.96 ms, 120 µs per bit). So the ratio between sending data and waiting time within the communication frame may not be the same depending on the game, and additionnaly, clock speed seems possibly faster that expected (see GB pandocs).

I worked with Arduino Uno.

So my theory (and possible conclusion) is that Arduino (at least Uno) is not fast enough to capture most of the game printing protocol, with the happy exception of GB Camera. It misses bits sometimes, ruining the communication protocol.

[minimal_GBPrinter.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/4574799/minimal_GBPrinter.zip)

mofosyne, 2020-07-28:
Update, I got half an image. I am suspecting its something to do with the status bytes.

[image]

Raphael-Boichot, 2020-07-28:
I think you will motivate me enough to plug my oscilloscope.

Glad to now you're still working on.

Regards.

Raphael-Boichot, 2020-07-29:
The double clock speed theory is true, I sent screen capture from
oscilloscope on Discord.

Raphaël

mofosyne, 2020-08-02:
Not sure if it's a speed thing, I've created a sniffer implementation in https://github.com/mofosyne/arduino-gameboy-printer-emulator/tree/master/research/gbp_sniffer which you can try out to grab all the packets from a working gameboy printer (that I recently got).

Here below is the capture i got from a gameboy pokemon cart to a real gameboy printer:

https://github.com/mofosyne/arduino-gameboy-printer-emulator/blob/master/research/gbp_sniffer/captures/2020-08-02_PokemonSpeciallPicachuEdition.txt

Raphael-Boichot, 2020-08-06:
Hello.

By using the version of code in attached document (one of yours, not
available online any more to my knowledge), if you use the "CHANGE"
interrupt mode, printing with Arduino is working in normal mode (GB
Camera), and if you switch to "RISING", it's OK in double speed mode (most
other games), but not with GB Camera any more. There is still rarely a
missing tile at the end of transmission in double speed mode. See Discord
for some results.

I have no idea why a setting working in double speed mode would not work in
normal mode (and inversely). I think there is a tricky issue with timing
coming from Arduino that leads to this strange behavior. Perhaps a solution
could be to wait a few microseconds after interrupt is fired to read the
bit. Typically what RISING is doing : reading the bit at the middle of
clock cycle.

It is very puzzling to me that the RINSING trick do not work with GB Camera
now...

Raphaël

mofosyne, 2020-08-16:
V2 should have solved this now. So closing now :)

closed 2020-08-16 by mofosyne