mofosyne/arduino-gameboy-printer-emulator #28: Print Error #02 in Game Boy Camera

slashinfty, 2020-05-26:
I have wired up an arduino nano, uploaded the program, and set the baud rate. When I attempt to print, I get "Print Error #02," which translate to "Universal Game Link cable is incorrectly connected."

mofosyne, 2020-05-26:
Join our telegram for some support. We've been getting regular feedback that it is working, so it's likely to be a wiring error on your end.

mofosyne, 2020-05-26:
Telegram invite https://t.me/gameboycamera

mofosyne, 2020-07-28:
Were you trying to print pokemon? If so was it only half an image you got? Just tested further and got something like

[image]

mofosyne, 2020-08-16:
I assume all is well now. Feel free to reopen this ticket. V2 is out now

closed 2020-08-16 by mofosyne