mofosyne/arduino-gameboy-printer-emulator #30: A few possible optimizations (Transfer Speed/Flushing to File or Serial)

HerrZatacke, 2020-07-12:
Hi there,
As promised, here my thoughts/questions regarding a few optimizations:

## Speed up transfer
As [this project](https://github.com/applefreak/esp8266-gameboy-printer) seems to do it, the actual transfer period (not the fake waiting time) is almost double as fast, which I do not really understand why.

## Allow other operations to happen "during" print
Is it possible to have code being executed in the `loop()` function for a longer period e.g. saving lot of data to flash or updating an attached display without causing the GameBoy to go to show `Error #2`.
So:
* would it be possible to "save" as much data in ram during receiving and after a certain time "flush" it to, let's say, flash memory or `Serial.print()` it and during that period keep the GameBoy "waiting" until the printer is ready again.  
As far as I understand, the internal RAM of an Arduino is even too small to store one image (2Kb vs 6Kb required), so the "flushing" would need to happen multiple times during receiving one image (if that's even possible)?
* If that is not possible, maybe there is another way to tell the GameBoy to "wait longer" until the printer is ready again? This would make it possible to at least "pause" between two images when using an ESP, so it can run it's operations.

Just another thought... would it be possible to send a "busy" signal to the GameBoy by just using some additional "electronic" parts? Where you would "turn on" an additional circuit that responds to the GameBoy's "inquiry" commands while the Arduino is busy?

## Get Other games except the camera working as well.
Maybe the issue has to do with compression?
[I found a description of the algorithm here](https://shonumi.github.io/articles/art2.html). Here is is also mentioned that Pokemon uses said compression.

mofosyne, 2020-07-27:
Just created a working WIP rewrite in https://github.com/mofosyne/arduino-gameboy-printer-emulator/commit/516f0e5319e6044c25b4b92c8812aa1feaf8b38d you might find it interesting to read and maybe suggest anything else I can do to make it easier for you to use.

It's still under work, but key point is:

* Uses circular buffer so you can continuously use it without interruption.
* Finally able to get checksum working, but had to disable it, as the circular buffer implementation i got so far doesn't allow for jumping back for redownloading. (I noticed that on checksum error it resends the last 650B data payload from the gameboy camera.
* Got rid of fake printing and streamlined everything
* Comment better hopefully.
* Will stop using printf and just print each character one by one for more space and speed savings. Hopefully this means I can start figuring out compression, or even decoding tiles right in the arduino (leading to simpler decoders in the PC).

Other arch difference:
* Instead of byte by byte psudo spi slave, I've made it so that it can read in 8bits or 16bits at a time. This is a little bit horrible to do in AVR interrupts, but I imagine no sweat in 16bit or 32bit cpus which you plan to use it in.
* Instead of `(1<<x)`, I use a sliding bitmask, which is hopefully more efficient.

Other consideration:
* I should maybe figure out a circular buffer that allows for temporary writes of unconfirmed packets, so that on a failed checksum, I can revert and rewrite that section. I thought of a simple ping pong buffer, but at `650B*2` a pop, its not a good use of space.
* I would like to have a way to make it easier to debug other weird games out there, so need to have a way to read out payloads directly in hex. So I can ask others to run it on their gameboy and send me the payload. This would also possibly allow for automated unit test vectors of the library for longer term maintenance. 

<details>

```
!"INIT"
!"NULL"
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF F8 FF E0 F0 C1 E0 80 C3 80 83 00 83 80 03
FF FF 7F FF 5F 3F 8F 1F 4F BF 3F 1F 0A 14 44 30
FF FF FF FF FF FF FF FF FF FF 3E FF 3C 18 D0 09
FF FF FF FF FF FF EF F7 F3 E7 76 E3 10 26 B4 00
FF FF FF FF FF FF FF FF FF FF E3 FF E3 41 01 C8
FF FF FF FF FF FF BF FF 3E 9F 1C 9E 0C 0E 0E 9C
FF FF FF FF FF FF 07 FF 03 07 51 23 78 73 FF 7E
FF FF FF FF FF FF FF FF FF FF 8D FF 04 0C 60 04
FF FF FF FF FF FF FF FF FF FF 93 FF 01 01 48 01
FF FF FF FF FF FF FF FF FF FF C7 FF C6 83 22 91
FF FF FF FF FF FF FF FF FF FF 5C FF 00 08 3B 00
FF FF FF FF FF FF FF FF FF FF 7F FF 3F 7F 1F 3F
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
81 03 00 83 80 83 81 C2 C0 E0 F0 F0 FF FF FF FF
C4 E1 0C 01 00 0C 0E 0C 1E 1F 7F 3F FF FF FF FF
CB C1 C9 C3 18 C1 0C 18 FD 3E FF FF FF FF FF FF
D1 A0 90 E0 36 80 12 26 AE 77 FF FF FF FF FF FF
D9 80 5E 81 DA 0D 60 01 EB 77 FF FF FF FF FF FF
1E 9C 1E 9C 0C 9E 0E 87 BF CF FF FF FF FF FF FF
F2 7F 79 72 50 22 03 06 3F CF FF FF FF FF FF FF
C2 04 60 04 00 64 00 04 CD 36 FF FF FF FF FF FF
0D C8 1D C8 1D C8 9C C9 EB DD FF FF FF FF FF FF
32 01 3C 03 34 9B C0 83 D3 EF FF FF FF FF FF FF
77 38 7B 30 70 33 70 30 BA 7D FF FF FF FF FF FF
3F 3F 3F 3F 1F 3F 3F 1F DF BF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
!"NULL"
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
3F 50 EE 01 55 00 F7 08 15 7F F4 0B 5E 01 FF 00
36 C1 56 BB 9D 77 22 DD 0B F5 BE 41 ED 17 E7 1C
74 83 BB C7 26 DF 3D FA BB 74 16 E8 95 40 EA 00
DA 7D 8E F9 A5 58 D8 20 F5 00 9F 00 5B 04 FE 01
[... 302 lines cut ...]
C0 8C C0 80 C1 83 DF 8F DF 8F BF CF DF FF FF FF
43 81 31 99 B5 18 B0 19 1B 81 A7 C3 E7 FF FF FF
84 02 60 32 7C 3E 78 32 30 02 4C 86 CE FF FF FF
D8 08 3A 11 18 10 49 10 80 49 5A EC FE FF FF FF
00 21 A3 13 07 13 E7 33 A2 11 78 31 79 FF FF FF
C1 E6 C0 E0 C0 E1 C7 E7 C3 E7 FF E7 FF FF FF FF
61 00 6B 46 EF C6 EF C6 EF C6 E6 CF EF FF FF FF
20 20 22 24 2E 24 2E 24 2E 24 BD 66 7E FF FF FF
C2 04 C0 4C C0 4C C0 4C CC 40 C5 62 E7 FF FF FF
0A 13 12 C3 13 02 EB 12 9B 42 38 12 30 F8 F8 FF
43 81 03 01 03 01 03 01 03 01 03 01 01 03 03 FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
!"DATA"
!"PRINT", num_of_sheets:1, num_of_linefeed_before_print:1, num_of_linefeed_after_print:3, palette_value:228, print_density:64
!"NULL"
!"NULL"
!"NULL"
!"NULL"
# ERROR: Timed Out
```

</details>

mofosyne, 2020-08-16:
V2 now completed and merged to master.

* Speed up transfer - print wait time reduced

* Allow other operations to happen "during" print - circular buffer and stripped down ISR done

* Get Other games except the camera working as well - more game supported now

closed 2020-08-16 by mofosyne