mofosyne/arduino-gameboy-printer-emulator #32: A new output format for GB printer emulator and decoder that handle compression

Raphael-Boichot, 2020-08-12:


Raphael-Boichot, 2020-08-12:
Hello.

I propose to change the output format for the gbp_dev in order to avoid useless output and line change (one line = one data packet). It ease the process of decrypting images from the stream. I propose also to decrease the timeout to 500 ms.

I propose an Octave code to decode the stream that can handle compressed data (run example code). The image splitting is made according to aftermargin wich is always 3 between images.

Enjoy !

[New_pixel_perfect_batch_decoder and GB Printer.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5064847/New_pixel_perfect_batch_decoder.and.GB.Printer.zip)

Raphael-Boichot, 2020-08-12:
The code can support Zelda DX, all the Pokémon games, SMB Deluxe, Harvest Moon 2 and of course GB Camera !

mofosyne, 2020-08-16:
Should be all done!

closed 2020-08-16 by mofosyne