mofosyne/arduino-gameboy-printer-emulator #33: List of game compatibility

Raphael-Boichot, 2020-08-13:
Hello.

Please find a list of games compatible with the Arduino game boy printer emulator dev code, with compression flag and serial speed (verified with oscilloscope), as well as the code used to test the games. Please include somewhere in the project.

Other games not included are too bad to deserve unlocking something or in japanese. 

Raphaël

[New_pixel_perfect_batch_decoder.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5069877/New_pixel_perfect_batch_decoder.zip)

[Compatibility list.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5069841/Compatibility.list.txt)

Raphael-Boichot, 2020-08-13:
Tony Hawk's Pro-Skater 2 also uses a special palette, please find the new list :

[Compatibility list.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5069897/Compatibility.list.txt)

Raphael-Boichot, 2020-08-17:
Hello.

Here is the Octave Decoder V2 that embed the CrapPrinter into it as a function. You will find also a library of tricky test cases with comment in order to programm the best decoder ever.

This version replace the other.

Regards

[Decoder test cases.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5086916/Decoder.test.cases.zip)

Raphael-Boichot, 2020-08-17:
See Discord for discussion about compatibility of the emulator V2 with Arduino. It works more or less depending on people/device.

mofosyne, 2020-11-22:
Ah, I wonder where we placed it. Maybe we should transfer it to the ReadMe doc now, since it's been out for a while and it sounds like no-one has complained yet. Maybe we can safely say that all games works now?

Raphael-Boichot, 2020-11-22:
Hello Brian.
I think everything is OK now. There's still a bunch of japanese titles to
assess but according to the pandocs we exhausted all the tricks of the
serial protocol, so I do not now what could go wrong.
On my side I do not encounter any bug on Arduino at least. I know people
using your code without any issue.
I will still be happy to help if you need a sniff or oscilloscope
confirmation.

Regards

Raphaël

mofosyne, 2020-11-22:
Nah, no need. I intend to help maintain this codebase still, so I'll just wait till someone complains. You did well enough.

closed 2020-11-22 by mofosyne