mofosyne/arduino-gameboy-printer-emulator #39: Issue with parse mode and compressed data

Raphael-Boichot, 2020-08-19:
Hello.

I discovered that the data sent by the emulator V2 in parse mode are not accurate for compressed games.

The capture ressemble one of your first attempt to embed decompression in the emulator. I think you've merged an old version of code for parse mode.

[Parse_Pokémon_Trading_Card.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5099143/Parse_Pokemon_Trading_Card.txt)

[image]

mofosyne, 2020-11-22:
If i recall, this issue is now fixed

Raphael-Boichot, 2020-11-22:
To what I remember, compressed mode was just not supported on Arduino for
memory/speed issue, but OK with ESP devices. This was the only issue
reported.

Raphaël

mofosyne, 2020-11-22:
Well if I understand correctly, raw mode has full support for compressed stuff. So one strategy could be to just force everyone to use raw mode instead.

Raphael-Boichot, 2020-11-22:
I do not remind exactly as I have no access to the discussion on the
Discord Game Boy Camera any more. I think you let parse mode for
compatibility reasons with some converters.

mofosyne, 2021-01-31:
Done, V3 now is raw mode by default.

closed 2021-01-31 by mofosyne