mofosyne/arduino-gameboy-printer-emulator #43: New version of Octave/Matlab decoder + CrapPrinter

Raphael-Boichot, 2020-08-23:
Dear Brian, 

Please find the last version of the Octave Decoder + fake Printer. I've added some capture tests made with your last code this afternoon. The Main code can automatically interpret parse and raw mode, compressed or not, and the CrapPrinter is now a simple output option for the Main code (set by default). You can place everything on the same folder. It replaces the current two folders printer + decoder.

I think you can clear the issues now !

Thanks again for your work !

Regards

Raphaël

Raphael-Boichot, 2020-08-23:
I forgot the files...

[Octave_Decoder_and_Fake_Printer_V2.zip](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/5114749/Octave_Decoder_and_Fake_Printer_V2.zip)

mofosyne, 2020-08-30:
Thanks. Now moved to https://github.com/mofosyne/GameboyPrinterPaperSimulation

closed 2020-08-30 by mofosyne