mofosyne/arduino-gameboy-printer-emulator #46: Use web serial to load data

DrSkunk, 2021-01-31:
Awesome project, printing the cable holder while writing this.

Chrome introduced the web serial in version 77 : https://www.chromestatus.com/feature/6577673212002304
We could use this to directly load the data from the Arduino into the decoding algorithm.

I built a proof of concept browser serial monitor a few months ago: https://github.com/drskunk/web-serial-monitor

mofosyne, 2021-04-04:
I'll be very interested to see where this leads to. Will certainly be a very useful addition, especially if chrome finally decides to integrate it.

Currently looks like an experimental standard so is likely subjected to flux, so if merging it, don't forget to make it modular enough to allow for easy removal/updating especially if there is a change to the chrome API.

Btw how the latest version going? Another person with a uno mentioned something about timeout issues.

mofosyne, 2021-08-13:
Looks like it's baked for the raspberry pi pico. https://github.com/stacksmashing/gb-link-printer Nice work stacksmashing!

mofosyne, 2021-08-16:
[image]
tried your example with my arduino nano but got this on my chrome browser

mofosyne, 2021-09-03:
Should be in now

https://github.com/mofosyne/arduino-gameboy-printer-emulator/commit/346246dbb2010cbebd33439a2965ceee9974cb82

closed 2021-09-03 by mofosyne