mofosyne/arduino-gameboy-printer-emulator #48: Timeout error

hellroy, 2021-04-02:
Hi,

I am using an UNO (r3) and when I go to print it times out immediately with "Timed Out (Memory Waterline: 6B out of 650B)".

To get around this, I plugged in an actual Gameboy printer in-line and was able to both print and capture the data to the serial monitor without the timeout.

I am not sure if this has to do with my pinout or not, but will keep trying to solve the issue on my own.

Thanks for such a cool project. I am really exited about getting this to work!

Thanks,
Layne

mofosyne, 2021-04-04:
What's the output? If it did output the packets, I think it might be an erroneous error. If still got something out, try copying the hexes, but exclude the line containing the "Timed Out (Memory Waterline: 6B out of 650B)".

hellroy, 2021-04-06:
Hi @mofosyne! 

It turned out to be a pinout issue. I bought the cable on ebay and I think it was an offbrand version. The pinout that finally worked for me was based on this one I found on hackaday: https://hackaday.io/project/160329-blinky-for-game-boy/log/150762-game-link-cable-and-connector-pinout

Thanks again for the project. Super cool!

-Layne

mofosyne, 2021-05-23:
Glad to hear it all worked out. Don't forget to join our telegram chat and share some photos!

closed 2021-05-23 by mofosyne