mofosyne/arduino-gameboy-printer-emulator #49: Error compiling

slashinfty, 2021-05-16:
Downloaded v3.0.1 and received the following:
```
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/eightanaloginputs /tmp/build9666597091347293281.tmp/gbp_serial_io.cpp -o /tmp/build9666597091347293281.tmp/gbp_serial_io.cpp.o 
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/eightanaloginputs /tmp/build9666597091347293281.tmp/gbp_pkt.cpp -o /tmp/build9666597091347293281.tmp/gbp_pkt.cpp.o 
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/eightanaloginputs /tmp/build9666597091347293281.tmp/GameBoyPrinterEmulator.cpp -o /tmp/build9666597091347293281.tmp/GameBoyPrinterEmulator.cpp.o 
In file included from GameBoyPrinterEmulator.ino:95:0:
/usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:25:0: warning: "true" redefined
 #define true 0x1
 ^
In file included from gbp_serial_io.h:28:0,
                 from GameBoyPrinterEmulator.ino:35:
/usr/lib/gcc/avr/5.4.0/include/stdbool.h:46:0: note: this is the location of the previous definition
 #define true true
 ^
In file included from GameBoyPrinterEmulator.ino:95:0:
/usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:26:0: warning: "false" redefined
 #define false 0x0
 ^
In file included from gbp_serial_io.h:28:0,
                 from GameBoyPrinterEmulator.ino:35:
/usr/lib/gcc/avr/5.4.0/include/stdbool.h:45:0: note: this is the location of the previous definition
 #define false false
 ^
GameBoyPrinterEmulator.ino: In function ‘void setup()’:
GameBoyPrinterEmulator.ino:178:52: error: ‘digitalPinToInterrupt’ was not declared in this scope
GameBoyPrinterEmulator.ino: In function ‘void loop()’:
GameBoyPrinterEmulator.ino:210:19: warning: unused variable ‘sioWaterline’ [-Wunused-variable]
GameBoyPrinterEmulator.ino: In function ‘void gbp_packet_capture_loop()’:
GameBoyPrinterEmulator.ino:398:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
```

mofosyne, 2021-05-17:
[image]

Confirmed to compile on latest stable arduino IDE v1.8.15 on windows. What version IDE and OS are you using? Also looks like its failing on `stdbool.h` because the new arduino is redefining boolean values. Then try commenting out all `#include <stdbool.h>`.

Not sure how to fix this exactly.

slashinfty, 2021-05-17:
It appears the Arduino IDE in my stable repo was very out of date.

closed 2021-05-17 by slashinfty

mofosyne, 2021-05-18:
Glad to hear it's all sorted. Don't forget to join our telegram chat if you need any more help. And to share your photos!