mofosyne/arduino-gameboy-printer-emulator #52: Bug in two games of the Nakayoshi Cooking Series

Raphael-Boichot, 2021-06-24:
Hello Brian.

I've put two sniffs on the sniffer project with the Nakayoshi Cooking Series 1 and 2. The two games enter an infinite loop with the printer emulator because they expect a 0x04 status for continuing printing. In the current version, status command returns 0x00 what is not expected by these two games.

Hope you can fix the emulator easily based on these files.

Raphaël

[Nakayoshi Cooking Series 1 - Oishii Cake-ya-san_emulator.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/6711467/Nakayoshi.Cooking.Series.1.-.Oishii.Cake-ya-san_emulator.txt)
[Nakayoshi Cooking Series 1 - Oishii Cake-ya-san_sniff.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/6711468/Nakayoshi.Cooking.Series.1.-.Oishii.Cake-ya-san_sniff.txt)
[Nakayoshi Cooking Series 2 - Oishii Panya-san_sniff.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/6711469/Nakayoshi.Cooking.Series.2.-.Oishii.Panya-san_sniff.txt)
[Nakayoshi Cooking Series 2 - Oishii Panya-san_emulator.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/6711470/Nakayoshi.Cooking.Series.2.-.Oishii.Panya-san_emulator.txt)

mofosyne, 2021-07-31:
```
Raphaël BOICHOT — Yesterday at 9:59 PM
Kids are in a nap, time to make the printer suffer. I'm making some test to assess normal protocol and deviant protocol with my SD printer connected to a Pocket printer.
Normal behavior of the printer (COMMAND->printer response): 

INIT -> 0x00
PACKET#1 ->0x00
EMPT ->0x08 (Unprocessed data)
PRINT ->0x08 (Unprocessed data)
INQUY ->0x04 (image data full)
INQUY ->0x06 (image data full + printer busy)
...
INQUY ->0x00
Inserting INQUY commands at the beginning is OK :
INQUY -> 0X00
INIT -> 0x00
INQUY -> 0X00
PACKET#1 ->0x00
EMPT ->0x08 (Unprocessed data)
PRINT ->0x08 (Unprocessed data)
INQUY ->0x04 (image data full)
INQUY ->0x06 (image data full + printer busy)
...
INQUY ->0x00
BUT, if you insert an INQUY command in-between PACKET#1 and EMPTY packet, the state of printer is lost...
INIT -> 0x00
PACKET#1 ->0x00
INQUY ->0x00
EMPT ->0x00
PRINT ->0x00 
INQUY ->0x04 (image data full)
INQUY ->0x06 (image data full + printer busy)
...
INQUY ->0x00
If INQUY command is inserted in-between EMPTY packet and PRINT, the behavior is also weird :
INIT -> 0x00
PACKET#1 ->0x00
EMPT ->0x08 (unprocessed data)
INQUY ->0x04 (image data full)
PRINT ->0x04 (image data full)
INQUY ->0x04 (image data full)
INQUY ->0x06 (image data full + printer busy)
...
INQUY ->0x00
Conclusion : the INQUY command must not be used between the packet transmission and the print command, without resulting in weirdos printer response. Yet the five games with emulator issues use this protocol !
So the printer emulators should detect this unexpected (and i guess undocumented) use of INQUY :
- rule #1 : If INQUY is inserted between last data packet and EMPTY packet, the games expect a 0x00 response to trigger printing.
-rule #2 : If INQUY is inserted between EMPTY packet and PRINT command, the games expect a response 0x04 to trigger printing.
- rule #3 : Last case, if INQUY is used before and after EMPTY packet, rule #1 applies.
```

Raphael-Boichot, 2021-08-03:
Hello.

This file fixes the printer issue with the 5 last games and runs nicely with the others.

[gbp_serial_io.cpp.txt](https://github.com/mofosyne/arduino-gameboy-printer-emulator/files/6926901/gbp_serial_io.cpp.txt)

Raphaël

mofosyne, 2021-08-05:
Thanks. I will look into this and check at least that it works with gameboy camera before committing it in.

At least after copying over the compatibility list somewhere into here so we can keep track of what the current version can support. (So we can compare against your fix and avoid any regressions).

Raphael-Boichot, 2021-08-06:
Nice ! Thanks for your replay !

I've tested more than 30 games without any issue with the new version.

Raphaël

closed 2021-08-21 by mofosyne