mofosyne/arduino-gameboy-printer-emulator #58: Timeout

T0biasCZe, 2021-11-16:
The program always throws error "// Timed Out (Memory Waterline: 6B out of 650B)" while printing (and error #02 on screen)
Sometimes it does it after transfer
[image]

But sometimes the "print" screen flashes for half second and then it throws that error...
also once it threw it twice, one time mid transfer, and then after transfer finished

T0biasCZe, 2021-11-16:
example of console output when the printing screen flashes
[image]

(even though the timeout, the hex always works)

mofosyne, 2022-04-20:
Investigated. I can see that somewhere we dropped checks on the status byte when doing simulated printing for raw packet capture.

It's unlikely to cause any problem so I've just changed the message to "completed" rather than "timeout" which is more accurate anyway for https://github.com/mofosyne/arduino-gameboy-printer-emulator/releases/tag/v3.2.1

There may be more optimal changes, but to avoid breaking any other games that we worked hard to support, we opted to just do this small change.

Thanks for reporting this issue. Do let me know if you cannot download any images.

closed 2022-04-20 by mofosyne