mofosyne/arduino-gameboy-printer-emulator #71: Two adapter boards for your emulator

Raphael-Boichot, 2024-03-17:
Hi !

I've made two PCBs, one for Arduino Uno, one for Nano, compatible with your project. I've cross checked pinout with a working prototype. The Uno can also be used with some of my other projects, so the wiring for SD card.

https://github.com/Raphael-Boichot/Collection-of-PCB-for-Game-Boy-Printer-Emulators

Raphaël

mofosyne, 2024-03-18:
Nice job! I wonder where I could mention this in the readme. I'll try and figure it out or if you can push a change to it, that works too! Maybe there are some specific consideration you need to mention to make it work best with this project?

mofosyne, 2024-03-18:
Oh btw, is there a website or app in our community that supports this on android? Maybe a QR code for quick launch would be good to add. Just a suggestion

Raphael-Boichot, 2024-03-18:
Not to my knowledge. Mraulio on the Telegram may give you more information about that.